openbikesensor/portal #23: upload failing

uwewoessner, 2020-12-10:
I get the following error when I try to upload tracks to the current version of the obsAPI

CastError: Cast to ObjectId failed for value "None" at path "_id" for model "User"
    at model.Query.exec (/root/obsAPI/node_modules/mongoose/lib/query.js:4371:21)
    at model.Query.Query.then (/root/obsAPI/node_modules/mongoose/lib/query.js:4463:15)
Any idea what the issue could be?

opatut, 2020-12-11:
Where are you uploading tracks from? Can you give me an example file that fails (PM in Slack is ok too)? Do all files fail?

uwewoessner, 2020-12-11:
sorry user failure, The API key was wrong, it got lost during firmware update...

closed 2020-12-11 by uwewoessner

opatut, 2020-12-11:
Oh well, that's good to hear. Still, 

* the error message could be better
* and the API key should not get lost during update ;)