openbikesensor/portal #28: Stop using the user ID as a authorization token

opatut, 2021-02-17:
This is a real security concern right there:

* People are extracting their user ID ("api key") and putting that into the OBS device config.
* The OBS puts the ID into all files it creates.
* All requests to modify and fetch track data are made with that id as an authorization token.
* You cannot generate a new ID. It is permanent and used in the database for cross references.

Basically, we have to stop using it for two things (id + auth). We need a separate auth mechanism for the OBS.

There is already a JWT mechanism (implemented with *passport*). The angular app uses it. That token is a bit long though for use in the OBS, and it expires after 60 days. Not optimal, you'd have to reconfigure your OBS every other month. Here are a few options:

* For simplicity sake, we could add a token similar to the current one, one that is rather permanent, but that *can* be deleted or changed or recreated if needs be, on that *is not* identical with the user ID.
* Or we do use a JWT with passport and everything, but this token would be generated with different settings, e.g. much longer expiration date, and it could be scoped to allow uploads, and only uploads. This is somewhat harder to implement but would be considered "the proper way". Instead of showing the API Key on the account page, you'd get a button that says "request token for OBS", and when you click it, your token is generated and shown in a text box for copy&paste.
* Hardest but most seamless way of doing this would be a proper OAuth2 flow, such as *Implicit Grant*, with a token mentioned above, started and terminated in the Webapp served from the OBS itself. You'd click "log in to portal" on your phone, enter your portal credentials (directly into the portal, not into the obs web interface), then get redirected back to the OBS, which receives its token with that redirect. Again, this token can be long-lived and scoped, to match the OBS requirements and allow only what is necessary, but nothing more.
* For this kind of token that might get out of control of the user, JWT is sadly a poor choice. You cannot list the active login sessions in the user's accounts and allow to remove them manually, as their nature is to be decentralized through cryptographic signatures. We might want to switch to a different token type if we need that. We could still build a "invalidate all tokens" button (probably named "Log out from all devices") in the account page which sets some kind of flag on the user object that then does not match the generated tokens anymore -- but that is defeating the purpose of JWT being decentralized ;)

Most importantly, we have to start treating an auth token as what it is, and stop putting it into things like the data files which might get shared between people without them realizing they've compromised their whole track database.

amandel, 2021-02-17:
@opatut  - the OBS does NOT put the api-key/token into files it creates! The api-key is stored only in the configuration in the internal flash and might be "exported" via backup of the configuration. Which files do you refer to where the token appears in?

I would opt for a "random" token as today, but allow a generation of a new token on the portal side. At the time a new token is generated the old one must not be accepted any more. For now, I do not think that we need to support multiple tokens with different permissions or validity period - this might be added later if there is a need. This matches to your 1st option, I would even not add the possibility to "delete" or "change" the token. Just allow to create a new one.

opatut, 2021-02-18:
It used to be in the track metadata, or at least that was being discussed. That was a rather bad idea, of course. And the config file is still bad enough. Remember, if this ID gets stolen the attacker has full access to your account, forever.

I'd say we should go with the standard oauth approach and provide persistent tokens in the form of access/refresh tokens, which get a scope attribute, such that we can restrict their use api-side. For example, it would be enough to allow the obs track upload, but nothing else. That would be the safest option, if you steal my OBS and extract the token from the flash, you can... upload files into my account, but not download existing content.

For the OBS firmware, not much will change. The authorization header will look a little bit different, the tokens will look a bit different, but still be strings, and at some point we'll have to implement refreshing the token (i.e. using one "refresh" token at boot/start of upload to generate a second "access" token, which is then used in the following requests to perform the upload(s)). That way the access token can expire quickly (hours), while the refresh token is long-lived (years?) and can be managed through the user interface (e.g. "Log out from this device."). Again, all of this is portal-side code, with minimal firmware changes, and gives us lots of flexibility and options for additional security features.

amandel, 2021-02-18:
My understanding is the only permission you currently gain with the token is to upload tracks for a certain user and the flaw we need to fix asap is, that it is permanent and used as identifier at some other places?

It was in discussion to be placed into the metadata of the csv file, but as soon as it became clear that this is not only a user id but also for authorization we dropped it. 

I do not understand the current need for multiple tokens, with possible different permissions behind or different expiration periods per user. You might already have different use-cases in mind?

opatut, 2021-02-18:
It's just best practice. I want to have manual control over logging out my other devices, invalidating their tokens selectively, for example when I lose my OBS/ it gets stolen. I'd rather *do it right* now than provide a quickfix that will need iteration later.

OAuth 2.0 is industry standard for a reason. It allows all of these patterns that we'll build at some point. Why not do it once and for all?

> My understanding is the only permission you currently gain with the token is to upload tracks for a certain user

Sadly, no, it grants full access to the account.

> permanent and used as identifier at some other places

It is the other way around, it *is* the user ID, which is the identifier, but that cannot be changed. So we have to move the authorization to a different token. If we do, we can call it "access token" and do all the other OAuth2 stuff too, so it adheres to a standard everything and everyone can work with

It's not that complicated and not that different from a single token that is distinct from the user ID, only it's an unlimited number of tokens that you can create. And then we can attach more information to those tokens. We don't have to, but if we build it right, we can, even later.

> I do not understand the current need for multiple tokens

If I have multiple devices, for example. And also I want this to be the same mechanism for all login "sessions", so the web app should use the same (just get an unscoped token, compared to a restricted token which we store in the OBS).

Don't worry too much, it sounds complex, but it's not that bad. I'll build it some time soon ;)

amandel, 2021-02-18:
> Sadly, no, it grants full access to the account.

Oh, I was not aware of this. To login you still use username and password.

> If I have multiple devices, for example

You could share the same new "access token" for multiple devices. From discussions, the still unsolved use-case is, if a device is shared by multiple users.

> Don't worry too much, it sounds complex, but it's not that bad. I'll build it some time soon ;)

I miss the big advantage over a "access token" attribute that is attached to each user and can be regenerated on demand. I see the point, if you use the same mechanism for frontend <-> backend communication. Nonetheless, OAuth 2.0 brings in more complexity than a shared secret "access token".

opatut, 2021-02-18:
> You could share the same new "access token" for multiple devices.

Yes but I don't want to.

> shared secret

You just contradicted yourself ;)

opatut, 2021-02-18:
@amandel is there any restriction on what the token string can look like? Full utf-8, or will that break your serialization/JSON stuff/http parsing/...?

amandel, 2021-02-18:
@opatut - can you please tell me where I contradicted myself, looks more like we have a misunderstanding, or I was not clear?

I'd prefere something that does not need escaping, I do not know how far we can trust the used libraries. We do not have any kind of Unicode support on the ESP, but that should not be needed for a token? I assume the user needs to copy & paste it over?

opatut, 2021-02-18:
> can you please tell me where I contradicted myself

That was tongue-in-cheek :stuck_out_tongue_winking_eye: Something "shared" can never be "secret".

> We do not have any kind of Unicode support on the ESP, but that should not be needed for a token

I've been using a hex string now. No problem. It's a bit longer than the old User ID, but apart from that, it is very similar.

> I assume the user needs to copy & paste it over?

Until I build the auth code flow into the config server, yes. Again, it's going to be a hexadecimal string.

amandel, 2021-02-18:
OK, I did not get it, a password or auth token is a typical shared secret, opposed to public / private keys.

As long as we do not talk about kilobytes, there is no limitation in the length of the key, so it should be all fine. You mentioned it needs to be sent differently, so the portal might need to support both ways over a time, a forced OBS Update would also be an valid option. Recording of tracks is still possible, but to upload the data a firmware update is needed then.

opatut, 2021-02-18:
I'm taking it slow so probably no forced update will be necessary.

I've been reading up on oauth best practices and additional security considerations, and we should do them all. Seriously, authenitcation is rather important in a system that deals with potentially very personal data, so we must take extra care. Even sending the password from the webapp to the server in exchange for a token (called "implicit grant") is not considered secure enough anymore. See [1] and [2] and [3].

[1]: https://oauth.net/2/grant-types/implicit/
[2]: https://tools.ietf.org/html/draft-ietf-oauth-browser-based-apps-07
[3]: https://tools.ietf.org/html/draft-ietf-oauth-security-topics-16

opatut, 2021-02-26:
Okay, I've been preparing the `v0.2` branch with the whole oauth stuff. It's almost done, some UI cleanup and improvmenets are still missing. I have *not* removed the OBSUserId auth scheme, and the new React frontend still exposes the User ID as-is to the current logged in user (called `apiKey`) in their profile settings. I did not want to enforce any changes elswhere with the rewrite.

However, this means we *can* start working on and testing the changes in the OBS itself to make it more secure.

@amandel If you want I can try to build a very small piece of JS code that handles the OAuth process. For that we need

* a URL on the config server to the script in
* an endpoint on the config server that accepts a new token
* The uploader will have to exchange the refresh token for an access token before upload. That is *one* POST request ;)

Do you want to start this project sometime soon or is there more urgent stuff to do? We can't merge it anyway before we release the new portal, and nothing will break if we don't do this soon ;)

amandel, 2021-02-27:
Cool to see progress. I'm currently trapped in implementing the support of https for the config-server running on the ESP. I see this as a kind of a base to get better security (or security at all). https://github.com/openbikesensor/OpenBikeSensorFirmware/issues/225 - besides the work to be  done (the https server has a different api as the http server used before) I'm at 104% of the available flash memory - even after the initial easy catch cleanups. So this will take some additional time.

opatut, 2021-12-01:
> I would opt for a "random" token as today, but allow a generation of a new token on the portal side. At the time a new token is generated the old one must not be accepted any more. For now, I do not [...]

This is now implemented in 0.3.2

I don't think we're going to change the whole API Key logic anytime soon, so I'm closing this. We are however using API keys for one thing only: Uploading new tracks. That makes the attack surface rather small, as you can't even *read* user data with the API key from the device.

closed 2021-12-01 by opatut