openbikesensor/portal #29: Associate tracks to regions

opatut, 2021-02-17:
Let's provide a way for admins to manage regions, as a list of geo boundaries with names and IDs.

Then detect which track is in which region. This should of course be updated when the track changes, or when a region is edited.

I'm thinking of very high level regions for now (Stuttgart, Hamburg, Berlin, ...) and not too many overlapping/hierarchical regions.

This way we'll be able to later filter by region, eg. for exports, or simply for finding other users or similar tracks.

amandel, 2021-02-17:
Can this region areas be derived from some data already available in/via open street map? (Not sure if such a thing exists but I would expect this)

opatut, 2021-02-18:
Sure it does, e.g. here is the boundary of [Stuttgart](https://www.openstreetmap.org/relation/2793104).

I would not automatically create those, but we sure can make the UI for creating regions fetch this information, so you only have to search for e.g. "Stuttgart" and get a suggested region displayed that you just need to confirm.

amandel, 2021-02-18:
Cool, does this also include an hierarchy so that we could (auto) label the tracks with "city:Stuttgart" "state:Baden-Württemberg" "country:Germany" might be "county:Stuttgart" / "district:Heslach". No sure - I bet you had a different use case in mind.

opatut, 2021-02-18:
The problem with those regions is that a track can span multiple of them. That's why I don't want to find all global regions across hierarchy levels that contain a track, but rather just limit it to some OBS-ecosystem relevant regions, such as the metropolitan area (Stuttgart, Hambrug, Berlin, ...) for those that are active at all. That saves computational efforts to find all intersections between the track and the regions, because the number of regions will be much smaller.

opatut, 2022-01-19:
Related to #112.

closed 2024-07-25 by opatut