openbikesensor/portal #30: Add privacy zones

opatut, 2021-02-17:
Users should be able to edit their privacy zones.

* [ ] Add a database model
* [ ] Add CRUD routes (REST)
* [ ] Apply privacy zone filter when turning trackData into publicTrackData
* [ ] Write tests for privacy zone filter
* [ ] Build frontend to manage per-user privacy zones (basic functionality)

opatut, 2021-02-17:
I've assigned myself because I have a WIP branch. I might push it later when we're done merging repos and then unassign myself, for somebody else can then continue to work on that branch. Hit me up if you're interested.

amandel, 2021-02-17:
Not sure if you saw https://github.com/openbikesensor/OpenBikeSensorFirmware/issues/216 - I would still stick with our current solution and close the linked issue - might be you have a different opinion?

opatut, 2021-02-18:
Yes I saw it. I want to get started with circular regions first, and then expand it maybe later. I think it's a good idea, but I'd rather keep it simple for now. If we store the geometry as GeoJSON and use a GeoJSON library to evaluate the region in the filter, it'll be very easy to upgrade to other shapes (Polygons) in the future.

amandel, 2021-02-18:
So we keep the current definition of the privacy areas. On the portal side there there might be the possibility to define different shapes in the future.