openbikesensor/portal #35: Add privacy disclaimer

groma84, 2021-02-18:
Creating an account needs to show a privacy disclaimer that needs to be accepted by the user before tracks can be uploaded.

For this we need a privacy disclaimer text first, and then the corresponding components and database fields.

And we need to discuss how to handle existing users - maybe force the disclaimer on them the first time they login again?

opatut, 2021-05-15:
* Registration can only happen with accepting the privacy policy. There is already a field in the frontend config for a URL where the policy is retrievable. We have to add a mandatory checkbox with the corresponding text and a link to the registration page. 
* For exsiting users, we'll migrate to the new portal by having them sign up again, anyway. So we need no migration strategy.
* This also eliminates the need for additional database fields.
* For the actual content of the "official" portal instance, this does not really belong here -- but @tobst apparently is working on a text for that.

gluap, 2023-05-19:
As we do now depend on keycloak for login management, keycloak can also serve a "user agreement" and keep track of who signed it. On updates of the agreement, re-signing can be made mandatory.