openbikesensor/portal #36: Confirm source code license

groma84, 2021-02-18:
At the moment we are using the
GNU Lesser General Public License v3.0
for the source code.

We should briefly discuss if we want to keep this license or change to another one.

opatut, 2021-02-18:
Hi @groma84!

The source code in this repository is, until now, contributed only by the following people:

* @groma84 
* @uwewoessner 
* @opatut 

And also dependabot and the technical user over at the hlrs server, so probably @uwewoessner or @tobst.

So only us 4 persons need to agree if we want to change the license.

---

I'm not a big fan of GPL/LGPL, but I'd be okay with keeping it. I would prefer something more open still, such as MIT. I don't particularly like sharealike licenses. I'm against licenses that exclude certain usage (such as anti-military or anti-nuclear), even though I support the notion, I think making the software non-fit for such purposes is more useful than sueing the military if they decide to use it anyway.

amandel, 2021-02-18:
I did not contribute to the portal part of OBS but we should have one license for all sub-projects. 

I would also vote for a MIT license for the source code part of our project.

BTW: Should we add a "OBS" or "global" git repo just for general issues and planing, Does anyone has experience how other projects handle this?

Steffeng5, 2021-02-18:
You have to notice that the contributors you mentioned are the ones from this repo. 
The stuff you copied from the AngularJs App is also from other people you can not find in the commit history here :)

Maybe we should also look in the "old" repo for obsApp for other contributors.

PS: Okay I looked into obsApp: Only myself is missing here :-) So everythings okay, just add my name in your post above :D

I'm not so familar with these licences. What is the pro point on MIT over GPL?

opatut, 2021-02-18:
> The stuff you copied from the AngularJs App is also from other people you can not find in the commit history here :)

Which stuff? I did not copy anything from obsAPP, I wrote a new frontend using create-react-app.

opatut, 2021-02-18:
> I'm not so familar with these licences. What is the pro point on MIT over GPL?

* https://tldrlegal.com/license/mit-license
* https://tldrlegal.com/license/gnu-lesser-general-public-license-v3-(lgpl-3)

Basically, if you take MIT code and modify/derive from it, you don't have to put it as MIT. The same does not apply to GPL code. That's the "sharealike" clause. Thare are of course tons of other details, but those are IMO the most important onces.

MIT  is pretty "do whatever you want". GPL/LGPL is rather restrictive, it forces every user to stay in the open source environment. LGPL is the "lesser" version of that, allowing the use of the software without modification in the context of other, proprietary systems. Doesn't make too much sense for this software IMO though.

Disclaimer: I'm not a lawyer. This is no legal advice ;)

uwewoessner, 2021-02-18:
+One Vote for LGPL

opatut, 2021-02-18:
IMO, we're not at the "voting" stage yet, we're at the "discussing options" and "providing arguments for/against each" stage.

opatut, 2021-02-18:
Edited @Steffeng5 into the list because I forgot that his map visualization code was of course transfered, which is waay enough to consider him an author. Sorry, @Steffeng5.