openbikesensor/portal #39: Reject empty tracks

opatut, 2021-02-19:
One part of openbikesensor/obsApp#14 that I overlooked is where empty tracks should not show up at all. Those happen quite often when the OBS is switched on, but turned off again  before starting the recording (i.e. no GPS found yet).

I'd say the API should just detect and reject empty tracks on upload. Maybe in addition we can teach the firmware not to produce these files (or detect and delete them), @amandel?

amandel, 2021-02-19:
Empty files with no data or header data only should not be generated any more since a while now. You might get files with a low number of lines or with no line with location information.

opatut, 2021-05-15:
The importer should reject those. Then we can mark the track as invalid, and delete or at least hide it.