openbikesensor/portal #47: Store track points as GeoJSON

opatut, 2021-03-01:
We should store the track's points not as a weird selfbuilt format but as propert GeoJSON. That'll be easier to maintain in the future.

Steffeng5, 2021-03-01:
Haha your first text  triggered me to cry „nooooooo“ and then I saw you edited the initial message :)

i‘m not sure which geometry you want to use, cause that matters much for the visualisation and handling with Frontend map frameworks. Mostly we Need points for further processing and visualisation, partly combined to linestrings.

opatut, 2021-03-02:
At the moment the API process files into "TrackData" objects with this weird shape:

```javascript
{
  date: String,
  time: String,
  latitude: Number,
  longitude: Number,
  course: Number,
  speed: Number,
  d1: Number,
  d2: Number,
  flag: Number,
  private: Number,
}
```

That is all rather hard to convert, and follows no standard. It is used **only** for the frontend to display the track, nothing else (currently) depends on this format. So we can easily pre-convert that format to whatever the frontend wants for display. For example, we can generate `eventsLeft`, `eventsRight`, `track`, `measurementsLeft` and `measurementsRight` separately (as FeatureCollection or LineString or whatever fits) and just fetch those when required, without transformation. Saves a lot of computing and complexity in the frontend. Also, this probably makes it easier to later aggregate e.g. "eventsLeft" into a public data pool.

opatut, 2021-04-19:
This is part of the new data pipeline now. The obs-face scripts generate some JSON files, one with all measurements, one with only confirmed measurements, and so on.

These files contain GeoJSON Feature Collections, which are features of Point Geometry, with properties (distances etc.) on each Feature.

Of course this is only an intermediate format and not the original we save. The original is the file as uploaded, and can be reinterpreted later as needed.

closed 2021-04-19 by opatut