openbikesensor/portal #73: Spaces are added to file names in some cases

Natenom, 2021-05-12:
When editing a track the track title is sometimes for example "2021 01 05 T 14 44 44 d 1 a 8 obsdata" instead of "2021-01-05T14.44.44-d1a8.obsdata.csv.csv". Same for viewing a track. The "original filename" is still correct.

opatut, 2021-05-13:
This is on kind of on purpose. The filename is used to derive a "proper" looking title. For example, if you trackfile is called `ride-to-altdorf.csv`, it will automatically name it `Ride to altdorf`.

We should do two things to improve this:

* Reenable the automatic track title generation based on the `recordedAt` date, e.g. `Thursday evening ride`, after the track has been processed and we actually know the recording date.
* Detect if the filename contains a timestamp itself, such as in your example, and use that to generate a timestamp-based title immediately upon upload.

opatut, 2021-05-13:
Improved in 7aee81dceebb9684bb014ded7dfd121b728bafd3. Please test and reopen if it still doesn't work.

closed 2021-05-13 by opatut