openbikesensor/portal #87: Track download  `Content-Disposition` does not work correctly

amandel, 2021-06-26:
When I download the track "original" with Firefox or Chrome, the proposed Filename is "download"? I dont know why, headers I see:

```http
accept-ranges: bytes
access-control-allow-origin: *
cache-control: public, max-age=0
content-disposition: attachment; filename="2021-06-26T14.39.36-387c.obsdata.csv"
content-length: 152842
content-type: text/csv; charset=UTF-8
date: Sat, 26 Jun 2021 16:44:26 GMT
etag: W/"2550a-17a4921bc32"
last-modified: Sat, 26 Jun 2021 16:23:05 GMT
x-powered-by: Express
```

OBS device direct download works fine with expected filename
```http
Connection: close
Content-Disposition: attachment; filename="2021-06-26T14.39.36-387c.obsdata.csv"
Content-Length: 152842
Content-Type: text/csv
Server: OBS/v0.8-dev
```

opatut, 2021-09-12:
[image]

Cannot reproduce. Which version were you using? Please reopen if it is still an issue.

closed 2021-09-12 by opatut

amandel, 2021-09-18:
Thanks for checking - was likely a older version with no display of the portal version number ;)