openbikesensor/portal #90: Portal responds with 200 to post to /api/tracks when openbikesensor mistakenly posts to frontend

gluap, 2021-07-02:
I accidentally configured an openbikesensor with the frondend url instead of the backend url. 

**Observed behaviour:** Triggering an upload from the OBS will display an upload success. Frontend logs 
```frontend_1  | 172.19.0.2 - - [02/Jul/2021:20:19:05 +0000] "POST /api/tracks HTTP/1.1" 200 2440```
which makes me suspect that OBS assumes successful upload based on Status code 200 (OK), therefore I'm filing in portal.

**Actual behaviour**: OBS displays upload success but the track never arrives on the portal. I suspect in order to find out that it is posting to the wrong address, the OBS needs to get a non-200 response.

boldt, 2021-07-02:
It's a known issues, but not reported yet 👍

opatut, 2021-09-14:
It is solved since we've moved the api to `/api` and the frontend to `/`, so the new api will respond to `/api/tracks`, but the route should be `/api/api/tracks` instead, so it generates a 404 now. See: https://portal.openbikesensor.org/api/tracks

closed 2021-09-14 by opatut