openbikesensor/portal #99: Data poits behind map

Zeitungsleser, 2021-10-05:
The data points are rendered behind the map, so that they are invisible. In the file TrackMap.tsx the zIndex is set, but in the finished html the divs and canvas don't have a zIndex.

opatut, 2021-10-05:
Not sure if the `zIndex` in the layer is translated to a `z-index` in HTML/DOM, doesn't necessarily need to be. I'll check again, I thought `305850e` had fixed that.

opatut, 2021-10-05:
Wait, are you talking about the per-track view or the homepage? I don't think I've fixed it on the Homepage Map...

Zeitungsleser, 2021-10-05:
> 
> 
> Not sure if the `zIndex` in the layer is translated to a `z-index` in HTML/DOM, doesn't necessarily need to be. I'll check again, I thought [305850e](https://github.com/openbikesensor/portal/commit/305850ea637442bb0fb85e5567f0e757d322d89f) had fixed that.

From my point of view it is not translated.

Zeitungsleser, 2021-10-05:
> 
> 
> Wait, are you talking about the per-track view or the homepage? I don't think I've fixed it on the Homepage Map...

It's at the page where you can look at a specific track. For example: https://obs.verschwoer.haus/tracks/autoupload-2021-10-04t160413-8cbe-28854i

opatut, 2021-10-05:
Oh, the problem was only fixed for the gps track (blue line), not the recorded events. It's fixed now.

> > Not sure if the zIndex in the layer is translated to a z-index in HTML/DOM, doesn't necessarily need to be. I'll check again, I thought 305850e had fixed that.
>
> From my point of view it is not translated.

But it's not supposed to be translated. The whole map is rendered into a single `<canvas>` element in the DOM, so z-index is not translated to anything you could inspect in the browser (the layers don't have individual div or canvas tags). Instead, openlayers uses it internally to sort the order of draw operations, I guess, such that things with higher `zIndex` are drawn on top of things with lower indices.

Anyway, it works now ;) Thanks for the bug report!

commit 7bba46fa (2021-10-05): frontend: show event points with correct zIndex, too (fixes #99)
closed 2021-10-05 by opatut