openbikesensor/portal #118: Discuss and reconsider color-by-distance scale

opatut, 2021-11-30:
It is arbitrary right now and should probably be discusssssed and improved.

gluap, 2021-12-18:
Currently the color-distance style matches well with the 1.5 metre rule that is valid within city boundaries (zone=DE:urban) in Germany. Unfortunately this makes the map green on rural roads where cars are faster and a 2.0 metre rule applies - even if the mean overtaking distance is way below the limit.

On the long term the scale should become "ruralness"-aware, on the short term we might say goodbye to the "green-red" transition and instead use less suggestive colors.

opatut, 2021-12-20:
I agree with the fact that right now they are wrong. I think we can normalize the values based on the legal minimum, have a scale where 0 is 0cm and 1 is the minimum legal passing distance, then have a color ramp going from 0.5 or so to 1.5 or so. We'll only have to show the color maps in meters, distinct by rural/urban, because I think a normalized scale is not that easy to understand when reading the map.

We could also just have one scale and two rows of values next to it, labeled accordingly:

[image]

(I'd flip it probably, urban on top, rural on bottom, but this would be the general idea).

gluap, 2021-12-20:
So the Idea would be to use the normalized distance for the coloring and keep the raw distance values in the tile data for the information table?

I'd also be a fan of making the "rural" property visible long-term in the "segment" detail view.

I agree on piling this with other schema changes.

One quick-and-dirty way that wasn't mentioned yet would be to allow the user to switch between "2 metre rule colors" and "1.5 metre rule colors".

opatut, 2021-12-21:
> So the Idea would be to use the normalized distance for the coloring and keep the raw distance values in the tile data for the information table?

Yes, but not for the information table. That one works differently, and doesn't depend on tile data, it does the lookup directly in the database with a special request. But in general, yes, I'd like to keep the actual distance in the tile data, as it is the baseline information and people might want to use it directly for different visualizations etc.

> One quick-and-dirty way that wasn't mentioned yet would be to allow the user to switch between "2 metre rule colors" and "1.5 metre rule colors".

That sounds like a good idea to get a quick improvement by just adding a tiny UI toggle, and then later we can make it generic with a tile change. Would you want to implement this for now, @gluap?

gluap, 2023-05-19:
I think this is obsolete now. - We have a new color scale and we also have the distinction of rural <> urban

closed 2023-05-19 by gluap