openbikesensor/portal #136: Show My Tracks when clicking on Tracks

Ventusfahrer, 2021-12-05:
[image]

At the moment I am more interested in ***My Tracks*** than in a list of latest uploaded tracks which are visible to public. 
May be this will change in future, but then I would like to use additional filters on public tracks.

opatut, 2021-12-05:
How about we put "My tracks" on the right, near your avatar, then keep "Tracks" on the left the public list, and remove the tabs on top of the list (where your arrow is pointing)?

Adding customizable filters will be a different issue I guess.

klausgoe, 2021-12-05:
I Propse to add the entry "My Tracks" to the list OpenBikeSensor Map Tracks. Then you have 
OpenBikeSensor Map (public)Tracks My Tracks
and you can directly switch/click to your own tracks. That avoids context dependant switching of content.

Ventusfahrer, 2021-12-05:
my original thought was:

when I login to the portal I am primary focused on my tracks, therefore reduce the amount of actions to get to them.

This is my initial idea, when opening the issue:
[image]
after clicking on ***Tracks*** I get a list of My Tracks and on the second Tab the ***Tracks***. I have to click on Tracks to get the list of the Public Tracks them. Maybe one can also rename Tracks on this level to "Public Tracks".

The second variant is probably the one you supposed (?):
[image]
This would allow to remove the 2nd Tab-Level, fine idea, which I like.

The third variant:
[image]

puts ***My Tracks*** to the User-Menu. This looks logical, but takes two clicks and moving the mouse to get to my tracks.

opatut, 2021-12-06:
Yeah, I thought about the menu too, but as you said, that is a primary action and takes two clicks, so it's not really an improvement.

I don't like the first option due to the "Tracks" button changing meaning based on your login state. So let's go with the second one ("My tracks" in top menu on the right).

opatut, 2021-12-06:
[image]

closed 2021-12-06 by opatut

klausgoe, 2021-12-06:
Don't know wether it matters, but I agree ;-)

commit 25ec75e7 (2021-12-06): Show "My tracks" directly in main menu (fixes #136)