openbikesensor/portal #142: Segment direction is reversed

gluap, 2021-12-18:
The direction in the road analysis seems to be backwards.

The screenshots below show a rather steep downhill westward direction. But the analysis looks like the cyclists in westwards direction were much slower. On this segment of road one will easily go up to 10m/s without pedaling, so it seems that the directions are reversed.

eastwards (uphill) display, note: driver is displayed around 33km/h
[image]

westwards, (downhill) display note: driver is around 10 km/h
[image]

I checked on one or two other segments with downhill and these were consistent with the assumption.

opatut, 2021-12-19:
What a good catch. I was already wondering about a place near me.

opatut, 2021-12-19:
Here is an even better image of the issue:

[image]

* Events all go in south-east direction (see text alignment).
* They are assigned to east-side road half.
* This is called the "north-west bound".
* This [road segment](https://www.openstreetmap.org/way/288082483) goes southwards in OSM.
* Therefore, north-west would be the "reverse" direction, which is as it should be the second tab.
* Let's inspect the database:

```
obs=# select distance_overtaker, direction_reversed from overtaking_event where way_id = 288082483;
 distance_overtaker | direction_reversed 
--------------------+--------------------
               1.83 | f
               1.93 | f
               1.61 | f
               0.44 | f
```

Yes, they are imported incorrectly.

opatut, 2021-12-19:
I'm stupid, the events are imported correctly -- `direction_reversed = f` means they go forward, which is southwards. The visualization is wrong only :)

opatut, 2021-12-19:
Fun, same mistake done twice. And it was the same that I did above -- False = not reversed = forwards. That's confusing ;)

closed 2021-12-19 by opatut

gluap, 2021-12-20:
Can confirm that after fetching ``main`` and re-running the sql initialization the visualization renders fine (without reimport, as expected)

[image]

commit 810bd391 (2021-12-19): Fix direction of road segments (fixes #142)

BREAKING CHANGE: This changes the tile generation function, please
update your database with `prepare_sql_tiles.py`.