openbikesensor/portal #143: namespace conflict between portal and scripts

gluap, 2021-12-18:
Pulling the latest portal and recursive submodules, I get the following issue in the worker:

```
worker_1    | Traceback (most recent call last):
worker_1    |   File "/opt/obs/api/tools/process_track.py", line 6, in <module>
worker_1    |     from obs.api.db import connect_db
worker_1    | ModuleNotFoundError: No module named 'obs.api'
```
this seems to be true as indeed the scripts are referenced by the site-packages and they don't contain that module
```
root@c472c06bc0c2:/opt/obs/api# cat /usr/local/lib/python3.9/site-packages/obs.egg-link 
/opt/obs/scripts
```
however, a manual import from the default directory is successful:
```
root@c472c06bc0c2:/opt/obs/api# python3.9
Python 3.9.7 (default, Oct 13 2021, 09:00:49) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from obs.api.db import connect_db
>>> 
```
nevertheless I still think that upon running the python file we encounter a namespace conflict as both "scripts" and "portal" try register their respective obs directories upon pip install -e (as used in the dockerfile)

opatut, 2021-12-19:
No, that is fine, it is intentional if the packages cooporate. I made that on purpose, and that's why it only works with the `postgis` branch (which is supposed to be merged soon, but I'll have to fix the non-postgis related code in there first).

See here for how that works, with the namespace sharing: 

https://packaging.python.org/en/latest/guides/packaging-namespace-packages/

This is why `api/obs/__init__.py` contains a weird line and nothing else.

opatut, 2021-12-19:
I mean we could discuss which is prettier, but I think putting all obs code into `obs.*` is pretty cool.

closed 2021-12-19 by opatut

gluap, 2021-12-20:
Nah, didn't know about pep #420, with that in mind I agree that having it as ``obs.*`` is an elegant solution. Should have googled before making an issue out of it.