openbikesensor/portal #147: UniqueViolationError: duplicate key value violates unique constraint

boldt, 2021-12-20:
Uploading a track results in an exception:

> worker_1    | asyncpg.exceptions.UniqueViolationError: duplicate key value violates unique constraint "ix_overtaking_event_hex_hash"

What can be the reason?

# Log

```
portal_1    | INFO: 
worker_1    | INFO: Annotating and filtering CSV file
worker_1    | DEBUG: Importing: /opt/obs/data/tracks/dennis/xj4awic7/original.csv
worker_1    | DEBUG: file identified as format version 2
worker_1    | DEBUG: read 6005 measurements
worker_1    | DEBUG: deriving missing velocities
worker_1    | INFO: 6005 measurements processed, derived values for 801 measurements (speed: 800, course: 1)
worker_1    | INFO: Statistics:
worker_1    | total files:            1
worker_1    | total measurements:     6005
worker_1    | valid measurements:     6005
worker_1    | confirmed measurements: 99
worker_1    | time range:             2021-11-05 09:45:20+00:00 to 2021-11-05 11:25:22+00:00
worker_1    | continuous time:        5999.0
worker_1    | continuous distance:    21062.797717627145m
worker_1    | continuous segments:    4
worker_1    | DEBUG: Found 361 roads in tile (14, 8678, 5269).
worker_1    | DEBUG: Found 264 roads in tile (14, 8679, 5269).
worker_1    | DEBUG: Found 106 roads in tile (14, 8680, 5268).
worker_1    | DEBUG: Found 160 roads in tile (14, 8679, 5268).
worker_1    | DEBUG: Found 471 roads in tile (14, 8678, 5271).
worker_1    | DEBUG: Found 41 roads in tile (14, 8680, 5269).
worker_1    | DEBUG: Found 605 roads in tile (14, 8678, 5270).
worker_1    | DEBUG: Found 362 roads in tile (14, 8679, 5270).
worker_1    | INFO: Removed 0 invalid measurements
worker_1    | DEBUG: Applied 2 filters.
worker_1    | INFO: Removed 1624 measurements without distance, kept 4381.
worker_1    | INFO: Removed 4282 unconfirmed measurements, kept 99 confirmed.
worker_1    | DEBUG: Writing file /opt/obs/data/processing-output/dennis/xj4awic7/measurements.json
worker_1    | DEBUG: Writing file /opt/obs/data/processing-output/dennis/xj4awic7/overtakingEvents.json
worker_1    | DEBUG: Writing file /opt/obs/data/processing-output/dennis/xj4awic7/track.json
worker_1    | INFO: Import events into database...
worker_1    | INFO: Write track statistics and update status...
postgres_1  | ERROR:  duplicate key value violates unique constraint "ix_overtaking_event_hex_hash"
postgres_1  | DETAIL:  Key (hex_hash)=(fd1b506f36654b4e54d53dba5a61ca0e8b98468901c8ed50f6f577589cabd95e) already exists.
postgres_1  | STATEMENT:  INSERT INTO overtaking_event (track_id, hex_hash, way_id, direction_reversed, geometry, latitude, longitude, time, distance_overtaker, distance_stationary, course, speed) VALUES ($1::integer, $2::varchar, $3::bigint, $4::bool, ST_GeomFromGeoJSON($5), $6::float, $7::float, $8::timestamp, $9::float, $10::float, $11::float, $12::float) RETURNING overtaking_event.id
worker_1    | ERROR: Failed to process track. Will continue.
worker_1    | Traceback (most recent call last):
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/dialects/postgresql/asyncpg.py", line 420, in _prepare_and_execute
worker_1    |     self._rows = await prepared_stmt.fetch(*parameters)
worker_1    |   File "/usr/local/lib/python3.9/site-packages/asyncpg/prepared_stmt.py", line 168, in fetch
worker_1    |     data = await self.__bind_execute(args, 0, timeout)
worker_1    |   File "/usr/local/lib/python3.9/site-packages/asyncpg/prepared_stmt.py", line 233, in __bind_execute
worker_1    |     data, status, _ = await self.__do_execute(
worker_1    |   File "/usr/local/lib/python3.9/site-packages/asyncpg/prepared_stmt.py", line 222, in __do_execute
worker_1    |     return await executor(protocol)
worker_1    |   File "asyncpg/protocol/protocol.pyx", line 201, in bind_execute
worker_1    | asyncpg.exceptions.UniqueViolationError: duplicate key value violates unique constraint "ix_overtaking_event_hex_hash"
worker_1    | DETAIL:  Key (hex_hash)=(fd1b506f36654b4e54d53dba5a61ca0e8b98468901c8ed50f6f577589cabd95e) already exists.
worker_1    | 
worker_1    | The above exception was the direct cause of the following exception:
worker_1    | 
worker_1    | Traceback (most recent call last):
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1802, in _execute_context
worker_1    |     self.dialect.do_execute(
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 732, in do_execute
worker_1    |     cursor.execute(statement, parameters)
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/dialects/postgresql/asyncpg.py", line 457, in execute
worker_1    |     self._adapt_connection.await_(
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/util/_concurrency_py3k.py", line 76, in await_only
worker_1    |     return current.driver.switch(awaitable)
[... 101 lines cut ...]
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1640, in execute
worker_1    |     ) = compile_state_cls.orm_pre_session_exec(
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/context.py", line 298, in orm_pre_session_exec
worker_1    |     session._autoflush()
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 2226, in _autoflush
worker_1    |     self.flush()
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3348, in flush
worker_1    |     self._flush(objects)
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3444, in _flush
worker_1    |     flush_context.transaction = transaction = self.begin(_subtrans=True)
worker_1    |   File "<string>", line 2, in begin
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/util/deprecations.py", line 309, in warned
worker_1    |     return fn(*args, **kwargs)
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1322, in begin
worker_1    |     trans = self._transaction._begin(nested=nested)
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 629, in _begin
worker_1    |     self._assert_active()
worker_1    |   File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 601, in _assert_active
worker_1    |     raise sa_exc.PendingRollbackError(
worker_1    | sqlalchemy.exc.PendingRollbackError: This Session's transaction has been rolled back due to a previous exception during flush. To begin a new transaction with this Session, first issue Session.rollback(). Original exception was: (sqlalchemy.dialects.postgresql.asyncpg.IntegrityError) <class 'asyncpg.exceptions.UniqueViolationError'>: duplicate key value violates unique constraint "ix_overtaking_event_hex_hash"
worker_1    | DETAIL:  Key (hex_hash)=(fd1b506f36654b4e54d53dba5a61ca0e8b98468901c8ed50f6f577589cabd95e) already exists.
worker_1    | [SQL: INSERT INTO overtaking_event (track_id, hex_hash, way_id, direction_reversed, geometry, latitude, longitude, time, distance_overtaker, distance_stationary, course, speed) VALUES (%s, %s, %s, %s, ST_GeomFromGeoJSON(%s), %s, %s, %s, %s, %s, %s, %s) RETURNING overtaking_event.id]
worker_1    | [parameters: (8, 'fd1b506f36654b4e54d53dba5a61ca0e8b98468901c8ed50f6f577589cabd95e', 5837597, False, '{"type": "Point", "coordinates": [10.679319658946316, 53.865514516371825]}', 53.865514516371825, 10.679319658946316, datetime.datetime(2021, 11, 5, 10, 9, 37), 1.1, None, 1.428002388575906, 2.0277777777777777)]
worker_1    | (Background on this error at: https://sqlalche.me/e/14/gkpj) (Background on this error at: https://sqlalche.me/e/14/7s2a)
```

opatut, 2021-12-21:
Yeah, I meant to address this at some point. Here is the explanation:

We have to protect the database against duplicate data, because that would skew the database. There are a few things that can cause the same event to be added to the database twice.

1. You reupload a file of yours (manually or due to OBS bug or whatever).
2. Someone else uploads a file of yours.
3. You edit a file a little and then upload it again.

The file-uniqueness validation only works for (1), as it hashes the file and ensures uniqueness *per author*. This is on purpose, as you might want to send a file to some other person for them to be able to analyze it. (2) and (3) have no mitigation, except disallowing duplicate events.

Therefore we create a unique hash of `(latitude, longitude, timestamp)` for each event. I assume no two people will accidentally press the button twice in the same location and at the exact same millisecond ;)

This of course also triggers e.g. if the event is at (0, 0, 0) or so. I'll need to do some work on that to filter out these false positives -- and then also have a proper error handling instead of just breaking the import ;)

boldt, 2021-12-21:
I can reproduce the error on an empty portal, e.g., I deleted the postgres, the api-data and the tiles.

1) Login with a user
2) Upload a track --> no error in the log
3) Set the track as public --> the errors in the log occur

It seems, that re-proecessing the to be public throws the **UniqueViolationError**

opatut, 2021-12-24:
That of course should not happen :D Thanks for investigating, I'll have a look. Probably just a SQL transaction that wants to be committed before recreating items ;)

opatut, 2022-02-19:
It doesn't happen anymore, I think I fixed it some time ago.

closed 2022-02-19 by opatut