openbikesensor/portal #149: Documentation and longterm architecture review

notudu, 2021-12-23:
- A review of terms used in the documentation seems neccessary. Unification will help to understand better.
-- developent of a valid way to setup the portal
--- Setup of keycloak works, but then it comes to errors when creating the database
-- A sketch layout of the architecture would help.
- An interface between development and it-operation should be implemented
-- both sides need to communicate
- don't rely on Docker, how about splitting the components in separate units 
-- (Reverse/HA-Proxy, Frontend Web, Keycloak, DB [Cluster])

gluap, 2021-12-23:
Hi @notudu,

Upfront: I don't think this is a proper issue - rather it's a list of points that you do not like about the project, its documentation and design choices. Issues are for constructive criticism, feature requests and bug descriptions, but this one sounds like "I'd rather you'd do everything (or most things) differently, go find out yourself how to do that". With that in mind let me still answer your points individually.

> A review of terms used in the documentation seems neccessary. Unification will help to understand better.
> -- developent of a valid way to setup the portal

Last time I went through the migration readme I was able to arrive at a working portal. After seeing your post about unspecific problems just now I tried again with the [Dev Readme](https://github.com/openbikesensor/portal/blob/main/README.md) and was successful starting from scratch with a dev setup - the only caveat being that I started the postgres container before keycloak, so I filed that as a pr. But as you suggest that you already have a keycloak container up and running you seem to have found that out by yourself, also the error messages are pretty self-explanatory. With another setup I once made the mistake to have a local postgres server running and mistakenly tried to forward that same port to the postgres container. But frankly trying to run two services on the same port on the same IP that was entirely my mistake as an operator.

> --- Setup of keycloak works, but then it comes to errors when creating the database

What errors messages/callstack, from which application? Which step do you paraphrase as "creating the database"? 

> -- A sketch layout of the architecture would help.

@opatut linked that one in the forum, the full layout is also documented in the docker-compose files.

>  don't rely on Docker, how about splitting the components in separate units
-- (Reverse/HA-Proxy, Frontend Web, Keycloak, DB [Cluster])

That's (one of) the points of using docker and compose: The setup **IS** split into separate units, actually the ones you name, and the separation into containers is much stronger than when running individual services on the same machine. Asking for removal of docker to split the services leads me to suspect that you may not have had much experience with docker-compose yet. If you desire you can use tools from the docker ecosystem to run individual services on different hosts, or replace some or all by baremetal deployments that you set up by hand. (for instance if you run a postgres db cluster or a keycloak already you can plug in those instead of the ones from the deployment documentation).

> - An interface between development and it-operation should be implemented
-- both sides need to communicate

I am mostly IT operations in the context of this project - the primary reason I came to the OBS community was because I had to run a portal. Whenever I approached developers with a politely phrased issue description (error message, what steps had led to it, etc...) they were very polite and responsive. And that despite the fact that half the time my tendency to overlook lines in the readme was the cause of the issue. I see that as a sign of a well-working communication between development and operations. You will also find this evidenced in issues and pull requests here on github as well as discussions on the forums.

My suggestion for how to approach setting up a portal are as follows:

- If you encounter an error message or problem that you cannot solve on your own using the documentation and a bit of google: You can ask in the forum. In order for people to be able to help you they will require details - the respective error messages, what steps you took leading up to it and may ask for extra bits of information as needed.
- If you encounter an error message that you can solve on your own it is considered polite to file an issue or PR on github or - if you do not feel comfortable with github - describe the issue and solution in sufficient detail on the forum for somebody else to reproduce the issue and your fix it for others too.
- If you feel that something is lacking (for instance writing and maintaining documentation on how to setup on baremetal instance) and you see that the community does not have the manpower to provide it, you are very welcome to talk to people, chances are that they welcome it if you want to jump in and create what you found missing. That's what open source is for.

As the original issue is ill-specified I'll leave it open a bit for further discussion but at some point close it (with the suggestion to split it into individual issues what can substantiate with more detail).

notudu, 2021-12-24:
All the best. Bye

gluap, 2021-12-24:
Closing this one for now as implied by @notudu. Feel free to reopen for discussion or create new issues with sufficiently specific content to address them.

closed 2021-12-24 by gluap