openbikesensor/portal #155: Document result codes and possible content if track upload fails

amandel, 2021-12-31:
Might be a link to the source code would be enough documentation :)

With https://github.com/openbikesensor/OpenBikeSensorFirmware/issues/193 we started the idea to return the link to the new uploaded track as body of the response. This can allow easy navigation to the track after upload.

Today there are more none technical errors detected on the portal which deserve a error code like `409` if the track is already uploaded etc. Might be this is already the case, I did not look into the code ;).

gluap, 2022-01-02:
Digging a little my result is: currently we're returning Code 400 (InvalidUsage) for duplicate tracks. We're not overly specific in what we return yet. This is the implementation of ``/tracks`` on the server side:

https://github.com/openbikesensor/portal/blob/ee6163b3013f3dd6c96e178123ebe85ccd438267/api/obs/api/routes/tracks.py#L90

opatut, 2022-01-19:
In principle, any code outside the 200s range is an error. The shape of the error response body doesn't make me happy yet so please try not to use it ;)

The response is a JSON encoded track object, so it has its own slug in there and you can build the URL yourself. Isn't that enough?