openbikesensor/portal #192: QueuePool limit

SirTificate, 2022-02-08:
I've recently upgraded from portal 2.x to 3.x using the docker deployment strategy.
After Uploading a couple of tracks and browsing the map the portal only returns errors. The reason is the set limit of the queue pool
`Traceback (most recent call last):
  File "handle_request", line 62, in handle_request
    from sanic.models.futures import (
  File "_run_request_middleware", line 28, in _run_request_middleware
    Awaitable,
  File "/opt/obs/api/obs/api/app.py", line 182, in load_user
    await req.ctx.db.execute(select(User).where(User.id == user_id))
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/ext/asyncio/session.py", line 211, in execute
    return await greenlet_spawn(
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/util/_concurrency_py3k.py", line 134, in greenlet_spawn
    result = context.throw(*sys.exc_info())
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1691, in execute
    conn = self._connection_for_bind(bind)
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1532, in _connection_for_bind
    return self._transaction._connection_for_bind(
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 747, in _connection_for_bind
    conn = bind.connect()
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/future/engine.py", line 406, in connect
    return super(Engine, self).connect()
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 3204, in connect
    return self._connection_cls(self, close_with_result=close_with_result)
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 96, in __init__
    else engine.raw_connection()
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 3283, in raw_connection
    return self._wrap_pool_connect(self.pool.connect, _connection)
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 3250, in _wrap_pool_connect
    return fn()
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/pool/base.py", line 310, in connect
    return _ConnectionFairy._checkout(self)
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/pool/base.py", line 868, in _checkout
    fairy = _ConnectionRecord.checkout(pool)
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/pool/base.py", line 476, in checkout
    rec = pool._do_get()
  File "/usr/local/lib/python3.9/site-packages/sqlalchemy/pool/impl.py", line 134, in _do_get
    raise exc.TimeoutError(
sqlalchemy.exc.TimeoutError: QueuePool limit of size 5 overflow 10 reached, connection timed out, timeout 30.00 (Background on this error at: https://sqlalche.me/e/14/3o7r)`

I guess the limit is too low. Can it be configured in the config.py or in the docker-compose.yml? It keeps crashing frequently in the current state. A restart of the portal cotainer only lasts for a couple of hours.

gluap, 2022-02-08:
I tried to add the configurables in the [poolsize-fix](https://github.com/openbikesensor/portal/tree/poolsize-fix) branch - does your setup allow you to test that branch to see if that helps? The new config variables to be added to ``config.py`` are ``POSTGRES_POOL_SIZE`` and ``POSTGRES_MAX_OVERFLOW``. I set them slightly above the default in response to this issue, but you may want to tweak them to your setup.

If it helps we could merge the branch.

SirTificate, 2022-02-09:
That was quick, that you!
I've just deployed the new branch and told everyone to hit the portal and try to break it.
I'll let you know how it went later!

SirTificate, 2022-02-09:
It might be a bit early to call it a success, but so far the error didn't occur again.
I've doubled the values and browsed the map while someone else uploaded a few tracks. The portal is still working as expected. I'm also pretty sure that if the error appears again it'll be sufficient to raise the values even more.

Thanks again! 👍

opatut, 2022-03-07:
Still happens with raised limits.

gluap, 2022-03-07:
Mist aber auch. In meinem Sandkasten wars gefixt, aber reallife deployment ist immer noch mal anders. Vielleicht könnten wir den Default "show roads without data" auf "aus" setzen? Ich hatte bei den Tests den Eindruck, dass die vielen road segmente beim zoomen besonders reinhauen - Wenn man mehrer Level auf einmal zoomt werden immer gleich ne ganze Stange kacheln gecanceled.

gluap, 2022-03-07:
https://docs.sqlalchemy.org/en/14/core/pooling.html#dealing-with-disconnects [pool_pre_ping](https://docs.sqlalchemy.org/en/14/core/pooling.html#dealing-with-disconnects) might be another thing we could look into (mentioning it lest I forget it again)

opatut, 2022-03-08:
Ich würde eher eine async semaphore auf die tiles route tun die max. poolsize/2 requests gleichzeitig abarbeitet... dann bleibt platz für andere requests, und nur die tiles werden langsamer bei Last.

gluap, 2022-03-08:
Wenn das canceln an sich das Problem ist, reduziert beides (default no empty roads und semaphore bottleneck) nur die Rate mit der sich der pool mit kaputten connections füllt. Default no empty roads vielleicht etwas mehr.

gluap, 2022-04-11:
Ich hab das Problem seit der änderung auf unserem Portal nicht mehr beobachten können auch nicht unter last.

hier erwähnt jemand, dass es auch eine Folge von ``debug=True`` beim Flask start sein könnte - haben wir das evtl versehentlich noch von einer debuggingsession an?

https://stackoverflow.com/questions/16636067/flask-sqlalchemy-timeouterror/16652345

gluap, 2022-04-11:
Ugly workaround meinerseits:

- für den Portal container ergänzt in ``docker-compose.yml``:
```
    labels: # traefiklabel brauchts natürlich auch weiterhin wenn benutzt
        - AUTOHEAL=true
    healthcheck:
      test: ["CMD", "curl", "--fail", "http://localhost:3000/api/mapdetails/road?longitude=0.0&latitude=0.0&radius=100"]
      interval: 20s
      retries: 2
      start_period: 30s
      timeout: 10s
```

Und am Ende der Compose - achtung: Container mit zugriff auf den docker socket. Ich hab mir diesen willfarrel angeschaut und entschlossen ihm zu vertrauen. Für Traefik machen wir das ja ähnlich.:

```
  autoheal:
    restart: always
    image: willfarrell/autoheal
    environment:
      - AUTOHEAL_CONTAINER_LABEL=AUTOHEAL
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
```

opatut, 2022-06-26:
0cbf03cd560d5e4033a9fe5cc3965572019184f3

I might have found it.

gluap, 2022-09-22:
I haven't seen this since @opatut s fix in june - Unless someone else has I think we could close it?.

opatut, 2022-09-22:
Me neither, I think we're good :+1:

closed 2022-09-22 by opatut

b12f, 2022-12-09:
I'm experiencing this on a fresh install. I've increased the pool limit to 128 and overflow to 64, which appeared to help for a while but ended up running into a wall of Timeouts and QueuePool limits again.

https://github.com/openbikesensor/portal/commit/0cbf03cd560d5e4033a9fe5cc3965572019184f3 is running, it does not seem to fix the issue.

PR #196 (2022-02-18): make pool_size and overflow configurable for worker and portal
Fixes #192