openbikesensor/portal #207: Change Colour coding for overtaking distance and show statistic distribution of events per street segment

rotger711, 2022-02-27:
All single events, average or median overtaking distances in street segments below 150 cm inside and 200 cm outside off towns should be marked at least in light red or orange. Only distances clearly above 150 cm in town (maybe above 175 cm out of towns) should be marked in yellow and green. 
My suggestion (in / out of town):
- below 100 / 125 cm - dark red
- 100 - 125 / 125 - 150 cm - red
- 125 - 150 / 150 - 175 cm light red / orange
- 150 - 175 / 175 - 200 cm yellow
- 175 - 200 / 200 - 225 cm light green
- above 200 / 225 cm green
I am aware that the sensors will not show overtaking distances far above 250 cm, so this is why the colour scale IMO could also be a bit less hard taking into account there could be some events that have not been measured. But no green below the STVO-Limit.
In addition I would find it very helpful to get back the bar graph that shows the distribution of overtaking events in a street segment.

opatut, 2022-03-13:
I would rather use a percentage of the legal limit as a scale, of you want to do that. Using your boundaries from the urban scale (100/125/150/175/200) we'd put that as 66%/83%/100%/116%/133%. Or we could round it a little to

* > 40% under: dark red
* 20%-40% under: red
* 0-20% under: yellow
* 0-20% over: light green
* > 20% over: green

Urban boundaries would be 90/120/150/180cm and rural would be 120/160/200/240cm. 

By using percentages, we make it independent of changes in the law (or different laws in different countries), *and* much easier to compute due to being normalized by a single value instead of having to provide two color ranges.

I think a 5-color scale makes sense, and we should use a discrete (stepped) color scale instead of a gradient, as it visually mixes in the map anyway, and is easier to look at and understand, IMO. Google maps traffic does the same.

opatut, 2022-03-21:
Added discrete colors in 1c52ce7de9ca9098dccec2f0e81479ca4dbcbd54. Still missing are considerations of the legal limit (derived from rural/urban for now).

PR #242 (2022-07-26): Implement difference between urban and rural for events and roads
@opatut This is my take at enabling the differentiation between "urban" and "rural" for the portal-based map.

It will require:
- schema upgrade (alembic) to add the "road" column to the overtaking_event table
- prepare_sql_tiles to change the tile schema (took me a while that dis determines what data is present in the tile)
- re-computation the events

Left for the future:
- Switching to the ruralness AI for categorizing non-tagged roads

Looks like this (note: border color on events conveys ``urbanity``, histogram redness thre [... cut]
closed 2022-07-22 by opatut