openbikesensor/portal #209: Add median 80 percentil

rotger711, 2022-02-28:
Input comes from Martin M from Stuttgart. To be able to show streets that are save for cycling add an evaluation of street segments with the median 80percentile. This means to show the median distance that ist overpassed bey 80% of all overtakers in one segment. Unlike the „normal“ median which shows the overtaking space that is kept by 50% of car drivers of all events, the median 80 percentile shows the overtaking space that is kept or exceeded in 80% of all events. In this way we can eliminate possible false low measurements and the very few very bad guys and still show where streets and bike lanes are comparably safe. On the other hand an infrastructure where some drivers keep the overtaking space but a rather high percentage of 21 - 49% of drivers doesn't, will still be marked as rather dangerous (which at least for less experienced bikers undoubtedly is true).

opatut, 2022-03-03:
The idea is good. Why not quartiles though, traditionally at 0/25/50/75/100%? I think this is again a case for experimentation. Since we provide single events in the export, associated with their `way_id`, someone could do some statistic analysis (group single event distances by way_id) and figure out which percentile would be a good number. This 80% seems guessed to me ;)

opatut, 2022-03-03:
FYI here is the fields we compute *at the moment* for each road segment:

https://github.com/openbikesensor/portal/blob/509e784521345b3ecd7c2204d895f828cf87c99e/tile-generator/layers/obs_roads/obs_roads.yaml#L6-L24

Those can of course be extended. Also, the `distance_overtaker_array` contains all distances for the way, so client-side percentile computation is also already possible on the road segment tile layer data (if your client supports it).

rotger711, 2022-03-03:
Agreed, I could try and do some tests for a few street segments. Is this possible with the exported data right now? The 80/20-split is not a guess but a method often used to evaluate an optimum result in means of effort and benefits in empirical studies.

opatut, 2022-03-08:
> Is this possible with the exported data right now?

It is, but it depends on your tools how that'd go.

You can probably import the GeoJSON into matlab or R and do your statistical analysis there, based on the attached way_id property of each overtaking even. You could also transform it to CSV and load that into Excel. Or maybe even QGis can do this. I have no idea how to do these things, but the data is there.