openbikesensor/portal #217: Logout does not work (keycloak session not invalidated)

gluap, 2022-03-10:
Currently the logout correctly logs out of the OpenBikeSensor portal, but does not invalidate the keycloak login.

Expected behaviour:
- log in
- log out via logout on openbikesensor page
- click login again
- asked for password and username again

actual behaviour:
- log in
- log out via logout
- click login again
- **logged into last logged in users session WITHOUT being asked for username and password**

The problem with this is that there is currently no way for an user to invalidate their session.

This means that for instance when logging in from third party system the next user of that system may be able to access the users data without having to provide credentials.

opatut, 2022-03-10:
There is a possibility of a so-called "backchannel logout" in keycloak, which is exactly this: The application, when ending the user session, sends the user agent to the auth server to terminate the session there as well.

It just needs to be implemented ;)

gluap, 2022-03-10:
Yeah, I'm currently looking through keycloak documentation.

opatut, 2022-03-10:
It might be that back-channel is actually the wrong name, I might be confused ;) https://openid.net/specs/openid-connect-backchannel-1_0.html

But there is a way ;)

PR #218 (2022-03-10): fix logout
this should work on instances that already permit redirect urls with astereisk - others need to add ``/logout`` to the permitted redirect urls.

I do not think this is especially elegant, but I my normal level of googling didn't turn up a more elegant reference way.

fixes #217
closed 2022-03-10 by opatut