openbikesensor/portal #228: Bring back histogram view per way

opatut, 2022-04-05:
Let's find a chart library that works well with our current frontend and isn't too heavy, integrate it, and then use it to bring back the per-way histogram of overtaker distances from the old visualization:

[image]

See also the [related forum discussion](https://forum.openbikesensor.org/t/visualisierung-ich-vermisse-die-saeulengrafik-fuer-einen-strassenabschnitt/727/3).

gluap, 2022-04-05:
In https://github.com/openbikesensor/OpenBikeSensor-Scripts/blob/main/visualization/roads.html we were using [ApexCharts](https://stackshare.io/stackups/apexcharts-vs-d3) which seems to allow formulating the chart quite concisely.

[d3js](https://d3js.org/) would be another classic hammer for this nail (albeit a slightly oversized and complex-to handle one).

boldt, 2022-04-05:
> isn't too heavy
> [d3js](https://d3js.org/) would be another classic hammer

I would take a library, which makes it quick and easy for us to implement. Yes, D3 is a hammer, but there are some React-D3-Wrapper for an easy usage. For example, I used **[recharts](https://github.com/recharts/recharts)** some years ago. I was very happy with it. We could use [BarCharts](https://github.com/recharts/recharts/blob/master/demo/component/BarChart.tsx) to implement a histogram.

One more advantage: Once we have the library, we can use it to implement other charts easily.

opatut, 2022-04-06:
I've started integrating apache echarts https://echarts.apache.org/en/index.html through https://www.npmjs.com/package/echarts-for-react. It seems to work fine. If you have a strong opinion about chart libraries tell me, otherwise I'll just continue with it because it's halfway done ;)

nosch2021, 2022-04-06:
Btw, small side note: I also circled the OBS Logo, because whenever I take a snapshot from the maps or graphs (for example for some slides) the Logo should be in, so that nobody forgets where the data came from :-)

opatut, 2022-04-06:
[image]

opatut, 2022-04-06:
Now I've also flipped the table to make it more intuitive without sacrificing scientific details, and renamed "Overtaker/Stationary" to "Left/Right" for now, as we're not using this in any right-hand traffic areas yet ;)

nosch2021, 2022-04-06:
I like the lower Graph with the flipped table the most. To me this a very easy explainable and therefore understandable Grapph and table to non-experts. Thank you for the quick response and quick proposed change.

Btw, the left and right column (sensors) is understood 19 left and 18 right is ok though I would expect the are normally equal. I wonder a bit about speed measurement with 24, would this not usually be 19 as well?

opatut, 2022-04-07:
The sensors don't always get a reading, but the speed is always there. So there were 24 events confirmd on this way, the left sensor saw 19 of them, and the right sensors had a reading in 18 of them. So apparently there is usually something to the right of the bicycle on this road while being overtaking (in this case it's "dooring zone" parking), and a few of the overtakings had no distance recorded, usually due to being very far (my sensor for example doesn't see anything > ~220cm) or very small (e.g. mopeds, or very small cars seen only "from above" while cycling on the raised sidewalk/"Hochbordradweg").

PR #230 (2022-04-24): Road details histogram
Fixes #228
closed 2022-04-24 by opatut