openbikesensor/portal #229: Upload directories

Zeitungsleser, 2022-04-06:
Add the posibility to upload a directory, that contains one ore more .csv file. Then all of the files in this directory should be uploaded.

opatut, 2022-04-06:
You can select multiple files at once, so you can go *into* the directory in the file selector and usually select all files with Ctrl+A, then upload them all at once. Does that do the trick?

Also, you can open your file browser (on Windows, it's called "Explorer" I think) and select many files, then drag and drop them onto the upload area. That should work just as well.

AFAIK there is no way of uploading whole folders in html forms, but I might be wrong.

Zeitungsleser, 2022-04-13:
Not exactly the problem I was facing... I have many directories, each with one .csv file. And for me it would be easier to select all the folders, and not go into each folder to select the file.

(I know with a little bit of scripting I might make a workaround, but I thought it would be a nice feature in general to select folders... So this Issue ins't very important, and if there is no (simple) solution, we can close it.)

boldt, 2022-04-14:
@opatut actually, it's possible to read a directory with JavaScript and then walk thru the files. I did it some years ago. I'm not sure, if it was a vendor specific feature. [This can be a pointer](https://stackoverflow.com/questions/3590058/does-html5-allow-drag-drop-upload-of-folders-or-a-folder-tree).

Maybe like this it is  possible to select a parent folder and walk thru the tree?

gluap, 2022-05-16:
One Trick would be to recursively search for files with ".csv" endings in explorer or similar and then drag the csv files from the find result to the uploading window.

opatut, 2022-07-28:
> So this Issue ins't very important, and if there is no (simple) solution, we can close it.)

> Im not sure, if it was a vendor specific feature

Closing this as "not worth the hassle of struggling with experimental APIs".

closed 2022-07-28 by opatut