openbikesensor/portal #240: Cumulative view of user tracks

gluap, 2022-04-30:
This issue is one of the feature requests from [this post](https://forum.openbikesensor.org/t/ich-wuensch-mir-ein-paar-neue-geaenderte-features/841):

Klaus writes:  Es gibt im Portal einen Knopf mit dem ich mir „meine“ Tracks anzeigen lassen kann. Ich fände auch einen Knopf mit denen ich mir „meine“ Messdaten auf der Karte anschauen könnte, hilfreich.

I spoke to him and the "nicest" rendering would of course be a map showing exclusively the events corresponding from the logged-in users own tracks.

I looked into the database and we have the data we would need for this filtering in the portal.

A possible side advantage would be comparing different drivers on the same roads - i.e. Klaus on his road bike vs me on my recumbent vs Anna on the cargo bike - But cooperation of the users would be required as this is obviously personal data, so only ones own data should be available.

opatut, 2022-04-30:
If we want to do this right, we implement some kind of filters into the tile generator, sucht that additional arguments are passed into the layer queries, and that we can filter e.g. by userId or time (see #241).

I believe for this, we'll have to override parts of [MvtGenerator](https://github.com/openmaptiles/openmaptiles-tools/blob/b8f0ae329f6e295a0abad741e0345e6a4abbaa85/openmaptiles/sqltomvt.py#L13), I don't believe dynamic arguments to the MVT function will be a feature ever supported by upstream openmaptiles-tools.

This sounds like a doable, albeit challenging, extension to the current system.

A quick solution would be not to implement this in the tile generator and instead use GeoJSON through a different pipeline, but then we'll have to implement all layers and their properties as both a tile layer query and in the GeoJSON pipeline, which is a lot of excess effort and will not always be in sync. Also, large user data collections will have the old performance problems again when rendering. I think the first approach is more desirable in any way, if we manage to pull it off technically.

opatut, 2022-05-01:
Hey, it was easier than I thought, overriding just one method and now we can add custom variables to our `getmvt` function :) I'll build this soon.

opatut, 2022-05-01:
[image]

PR #260 (2022-07-28): Dynamic tile arguments
Adds filters to the map to dynamically limit the data being parsed.

Fixes #240, #241.
closed 2022-07-28 by opatut