openbikesensor/portal #245: Config has no 'POSTGRES_POOL_SIZE'

boldt, 2022-05-15:
Following the [Deployment-README](https://github.com/openbikesensor/portal/tree/main/deployment) on a new system, I get:

```
$ docker-compose run --rm portal tools/upgrade.py
[...]
INFO: Running migrations...
INFO: Preparing SQL tiles...
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/sanic/config.py", line 113, in __getattr__
    return self[attr]
KeyError: 'POSTGRES_POOL_SIZE'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/obs/api/tools/upgrade.py", line 27, in <module>
    asyncio.run(main())
  File "/usr/local/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "uvloop/loop.pyx", line 1501, in uvloop.loop.Loop.run_until_complete
  File "/opt/obs/api/tools/upgrade.py", line 22, in main
    await prepare_sql_tiles()
  File "/opt/obs/api/tools/prepare_sql_tiles.py", line 49, in prepare_sql_tiles
    await import_sql(sql_snippets)
  File "/opt/obs/api/tools/prepare_sql_tiles.py", line 129, in import_sql
    app.config.POSTGRES_POOL_SIZE,
  File "/usr/local/lib/python3.9/site-packages/sanic/config.py", line 115, in __getattr__
    raise AttributeError(f"Config has no '{ke.args[0]}'")
AttributeError: Config has no 'POSTGRES_POOL_SIZE'
```

What is `POSTGRES_POOL_SIZE` and where does it come from?

boldt, 2022-05-15:
It was added in this MR by @opatut:

https://github.com/openbikesensor/portal/pull/225/files

boldt, 2022-05-15:
It means, [the default .env](https://github.com/openbikesensor/portal/blob/main/deployment/examples/.env) is missing at least two variables.

gluap, 2022-05-16:
The default values can be found in [the example config file](https://github.com/openbikesensor/portal/blob/87283476950ebaabea723238a28ad783ad8e8c42/deployment/examples/config.py). We should add them to the env file, too

commit 5e8830cc (2022-05-21): fix #245
closed 2022-05-21 by gluap