openbikesensor/portal #247: Lots of SQL errors

SirTificate, 2022-05-23:
I've been upgrading my portal install (docker) but constantly updating the git repo and rebuilding the containers.

Since a few weeks the worker doesn't process the tracks anymore. I thought I might have missed a few migrations in the past and tried to rerun them via `docker-compose run --rm worker tools/upgrade.py` but it crashes with the following error:

```sqlalchemy.exc.ProgrammingError: (sqlalchemy.dialects.postgresql.asyncpg.ProgrammingError) <class 'asyncpg.exceptions.DuplicateTableError'>: relation "road" already exists
[SQL:
CREATE TABLE road (
	way_id BIGSERIAL NOT NULL,
	zone zone_type,
	name VARCHAR,
	geometry GEOMETRY,
	directionality INTEGER,
	oenway BOOLEAN,
	PRIMARY KEY (way_id)
)
]
```
Then I looked into the logs of postgres and found these errors:

```
ERROR:  relation "road_usage" does not exist at character 13
STATEMENT:  DELETE FROM road_usage WHERE road_usage.track_id = $1::integer
ERROR:  current transaction is aborted, commands ignored until end of transaction block
STATEMENT:  DELETE FROM overtaking_event WHERE overtaking_event.track_id = $1::integer
ERROR:  relation "road_usage" does not exist at character 13

```
there are hundrets of these message. 

I suppose something is messed with my database, but how can I fix it?

opatut, 2022-06-08:
Sorry for being AFK for a while, I'll try my best to fix this soon.

It looks like you imported road data (from OSM), which means the `road` table exists (the osm2pgsql tool creates a new table every time it is run).

But then, the migration to create that table can't run anymore. This is not a problem if you do the normal setup order:

* Create table schema through all migrations
* Import data (overwriting `road` table)
* Later: Upgrade only new migrations, old migrations are not run again

Basically in your case, probably due to the upgrade order, the migration tool tries to run the `CREATE TABLE road` migration. That should not have happened if you followed this advice from [UPGRADING.md](https://github.com/openbikesensor/portal/blob/main/UPGRADING.md):

> 0.6.0: Make sure to upgrade to 0.5.1 first, by checking out that version tag and running migrations, then coming back to this version. This is required because the migrations have been edited to create the initial database schema, but if you run the 0.5.1 migrations first, your database will remember that it already has all the tables created. This is not required if you set up a new installation.

However, I believe that this might not be correct, and even if you followed that, it might not have worked as I designed it (though I tested it once or twice).

## Solution?

So far about the explanation what's going on, here is the fix: You can tell alembic to skip migrations. First off, we let it output which migrations there are:

```
$ alembic history
5d75febe2d59 -> a9627f63fbed (head), create table road_usage
9336eef458e7 -> 5d75febe2d59, create table overtaking_event
d66baafab5ec -> 9336eef458e7, create table comment
9d8c8c38a1d0 -> d66baafab5ec, create table track
35e7f1768f9b -> 9d8c8c38a1d0, create table user
920aed1450c9 -> 35e7f1768f9b, create table road
986c6953e431 -> 920aed1450c9, create enum processing_status
3856f240bb6d -> 986c6953e431, create enum zone_type
<base> -> 3856f240bb6d, create extensions
```

It looks like `920aed1450c9 -> 35e7f1768f9b, create table road` doesn't work for you, so let's migrate to the one before that, skip that step, and run the rest to `head`:

```
$ alembic upgrade 920aed1450c9
$ alembic stamp 35e7f1768f9b
$ alembic upgrade head
```

The *stamp* command basically makes alembic prentend that you're at the specified version with your current database schema, and it will run from there.

I look forward to hearing from you whether this solves the issue.

opatut, 2022-06-08:
That transaction stuff looks  indeed like something is in a broken state. Did you restart your database? Maybe try to stop the portal's API process and only run the worker, to see if that interferes? I'm not the best with database transactions though, there might be some problem in the code too.

SirTificate, 2022-06-17:
I think I might have fixed it.
What I did:
I stopped the Portal+Worker containers and rerun the update script. Afterwards I stopped everything `docker-compose down` and restarted all containers. Then I logged into Postgis and set all tracks to queued.
So far is worker is running fine and processing all tracks without any issues. Haven't tried uploading new tracks, but it's looking really promising so far.

SirTificate, 2022-06-17:
Ok, there are still some tracks failing, but with different errors. 
```
sqlalchemy.exc.PendingRollbackError: This Session's transaction has been rolled back due to a previous exception during flush. To begin a new transaction with this Session, first issue Session.rollback(). Original exception was: (sqlalchemy.dialects.postgresql.asyncpg.IntegrityError) <class 'asyncpg.exceptions.UniqueViolationError'>: duplicate key value violates unique constraint "ix_road_usage_hex_hash"
DETAIL:  Key (hex_hash)=(1e7f60c9a9c92b27d4dcbcb414af2b1f647abfc30799e6d893b9ee46c974cbed) already exists.
[SQL: INSERT INTO road_usage (track_id, hex_hash, way_id, time, direction_reversed) VALUES (%s, %s, %s, %s, %s) RETURNING road_usage.id]
[parameters: (328, '1e7f60c9a9c92b27d4dcbcb414af2b1f647abfc30799e6d893b9ee46c974cbed', 824230969, datetime.datetime(2021, 3, 26, 18, 48, 52), False)]
(Background on this error at: https://sqlalche.me/e/14/gkpj) (Background on this error at: https://sqlalche.me/e/14/7s2a)```
hope it's not a hash collision.

opatut, 2022-06-17:
Created as a new ticket #253.

closed 2022-06-17 by opatut