openbikesensor/portal #252: Road snapping snaps the track to rivers

Nugman, 2022-06-13:
It seems that the road snapping feature is snapping the track not only to streets but also to rivers:

See track https://portal.openbikesensor.org/tracks/5ej7q77a

[image]

There is no path along the river.

opatut, 2022-06-17:
> And as far as I know there is also no way to drive next to the river.

There is no river there :) I [looked it up](https://www.openstreetmap.org/#map=18/50.06260/8.25765) to check the tags of that river, and did not find any. Then it dawned on me:

The blue line is your original GPS track without snapping, the pink line is the track snapped to a road.

So even though I built that feature myself, I still got confused. Apparently, the blue line is misleading, and we should change it. I am thinking the same pink color, but dotted or dashed, and less width. Definitely not blue though ;)

Nugman, 2022-06-18:
LOL, that's a good one. Thanks for figuring out and apologies for the misleading bugreport.

Actually the location on the first screenshot has a river next to the road: https://www.openstreetmap.org/#map=17/50.12226/8.41784
So I assumed it might be the case in the second location also. Sometimes rivers are underground in urban areas ;-)

commit 4907f038 (2022-06-26): Make raw track not look like a river (fixes #252)
closed 2022-06-26 by opatut