openbikesensor/portal #272: Add attribution to OpenStreetMap

grischard, 2022-08-11:
As OpenStreetMap Foundation chairman and a cycling activist, I was double happy and proud to see that the openbikesensor map uses OpenStreetMap as a background. We are an open project that explicitly encourages this kind of use.

The use of the map data is in itself free of charge, although services based on this data may of course be charged for. What is legally required is that you include a simple reference to our project, for example "© OpenStreetMap” with a link to [openstreetmap.org/copyright](http://openstreetmap.org/copyright) in a map corner. You can find more information in [our attribution guidelines](https://wiki.osmfoundation.org/wiki/Licence/Attribution_Guidelines).

This is not an ego thing. It is necessary to bring more volunteers to create a better map, and we also owe it to our data suppliers. After all, they ask us to mention the use of their data, which we comply with via our copyright page.

Would it please be possible to add the necessary attribution to your map?

Thank you very much in advance.

vizsim, 2022-08-24:
Hi,
this might be a misunderstanding. 
being a osm-enthusiast myself and a co-host of one of the federated portals ( https://obs.adfc-brandenburg.de ) I am quite curious which map/portal you mean.
atm on our map the osm-attribution seems a bit hidden due to the "beta-banner", which is kind of a bug. but i guess there is no intention of leaving out the osm-attribution on purpose.

grischard, 2022-08-25:
Hi @vizsim 

Indeed, I do get the attribution when I scroll down! That's well-hidden, along with the footer - I thought it was simply a full-screen map.

The attribution guidelines require that attribution is shown without requiring interaction. I suppose this can get closed when the banner gets changed or removed.

opatut, 2022-09-12:
I apologize for the oversight, @vizsim is right that adding the banner on top of the application moved the map and broke the intitial viewport. Of course that needs to be amended. Thank you for the friendly notice, @grischard. 

Our community aiming to be *open* in all aspects as well, we're well aware that these things need to be taken care of and done properly. I'll get right to it and rectify the issue.

opatut, 2022-09-12:
I made the map smaller to fix the issue in the first place, and moved the attribution to the top to a) prevent this issue in the future, and b) fix the same problem on the track page, where resizing the map was no option. Review and release will follow within days, I hope.

commit c1c3797e (2022-09-13): Improve placement of map controls and popovers (fixes #272)
closed 2022-09-13 by opatut