openbikesensor/portal #280: Road usage ignores cycleways

gluap, 2022-09-22:
Currently we only import ways where we consider the possibility that a bicycle can be overtaken. This excludes cycling ways, and for the evaluation of overtaking events that is good.

Hovever to evaluate the road usage meaningfully we would sensibly consider all cyclable ways - otherwise we will not faithfully show where OBS have driven. Oftentimes this does not appear as a problem, but for instance here the "laerer landweg" is an integral part of the cycling infrastructure - its usage gets overlooked in the road usage because it is cycling/walking only.

[image]

opatut, 2022-09-22:
I am a bit torn here.

We should allow the track to be matched as "someone cycled here". Especially for these cases where we want to see whether people take detours to avoid bad roads.

However, if you press the button there, who should that count? What if there are service vehicles allowed and you happened upon one? What if people regularily use a road restricted for motor vehicles? I know at least one of these in my town.

I think outright ignoring these roads where cars shouldn't exist is not the solution. Likewise, ignoring roads where bicyclces *shouldn't* cycle is also not true to the real world. We might need to introduce a scoring system to guess what actually happened. In this case, nobody pressed the button, so it is more likely that the cyclist crossed the trunk road on the cycleway that is forbidden for cars.

gluap, 2022-09-22:
I agree - for button presses on cycling-only ways there is no easy solution. Long-term we might flag them as "ask the user before counting it". I think finding an automatic way answering the question "what did the user press the button for" is hard and the first step might be to ask people about specific button presses - maybe the openstreetmap is wrong (cycling way absolved from mandatoryness), maybe there's service vehicles - hard to tell just from a track, the openstreetmap and a button press.

But I'd claim that for the road usage allowing all roads that can be cycled will be mostly right.

opatut, 2022-09-26:
> allowing all roads that can be cycled will be mostly right

"can" is the key here, I'd say let's ignore motorways and state highways if those are generally restricted, and ignore the rest. In urban areas, people will always be cycling where they are not supposed to, such as service roads (e.g. bus and tram lanes), footpaths, and the like. Let's try to reduce matching to those ways if there are better alternatives though, e.g. if the bus lane is parallel to a normal road, then probably the cyclist used the normal road, even if the GPS track is more aligend with the service road.