openbikesensor/portal #330: Feature request: Delete all tracks

Nugman, 2023-03-21:
As a user I would like to be able to delete all uploaded tracks uploaded from my account. Currently it is only possible to delete individual tracks which is a tedious job when there are a couple of 100 tracks to delete.
Usecase: I moved my account to another OBS portal and would like to clear my account on portal.openbikesensor.org to prevent duplicates in case researches aggregate data from different portals.

gluap, 2023-03-21:
@Nugman I think this is a valid use case, and it will also make it easier to fulfill GDPR  duties of "I want to download/delete all my data".

But be aware that for the research use case the Tracks and Events can be deduplicated based on their hashes. In fact we already use this on the portal to prevent duplicate upload of the same track.

opatut, 2023-04-01:
This is implemented in `next` and will be in 0.9.0.

closed 2023-04-01 by opatut