openbikesensor/portal #355: Upload error on portal.openbikesensor.org

rai42, 2023-08-26:
I have a (largeish, ~3.8MB) track that fails to upload on https://portal.openbikesensor.org with the error message
`Track upload needs a single file in "body" multipart field`.

On https://portal.openbikesensor.hamburg the upload works, both portals show version v0.8.0 .
I'm using Firefox 116.0.3 .

Track CSV can be downloaded here: https://portal.openbikesensor.hamburg/tracks/dnyhjgnn

gluap, 2023-08-26:
@rai42 unfortunately you can only download the csv yourself, I can only download the GPX.

But nvm - I believe the error should now be fixed. The difference between the two portals is that we use nginx as a caching proxy on the central portal. I upped the limit on the possible upload size to 25 Megabytes now.

Could you try again?

rai42, 2023-08-26:
@gluap Yes, it works now, thanks!

closed 2023-08-26 by rai42