openbikesensor/portal #357: Select multiple street segments

Nugman, 2023-09-12:
Streets are segmented automatically from one junction to the next. In some cases on rural streets this leads to very short segments because of farm tracks. In real life I'd like to know the numbers for the whole length of the road - from one village to the next, in order to approach politicians to build a cycle path next to the road.
Therefore it would be nice if I could select multiple street segments with CTRL-Click.

[image]

gluap, 2023-09-12:
Actually we do not segment, but load the segments as-is from osm. Depending on who mapped and how this does mean very long segments (also not always desirable) or very short ones. So multiselection makes sense.

gluap, 2023-09-12:
There's already a request for this in #250

opatut, 2024-07-24:
Closed as duplicate.

closed 2024-07-24 by opatut