openbikesensor/portal #359: All my tracks stay in state "still beeing processed" and are not done even after days of waiting...

RedBike2000, 2024-02-03:
My first test drives with OBS worked well but after some days the tracks are not processed anymore. I tried already to delete all tracks and made a new upload but without success.

RedBike2000, 2024-02-03:
If someone wants to look after: My username is "RAL3003" and the tracks should be public. Thanks for any advice how to fix it.

gluap, 2024-02-03:
Thank you for reporting this - Seems I didn't follow my own upgrade instructions precisely enough resulting in a hanging worker (the process that processes the tracks). Provided you're on the central portal your tracks should now (or soon, depending on queue length) be processing.

RedBike2000, 2024-02-03:
Works well. Thanks for this quite quick fix.