openbikesensor/portal #360: Road direction mixed up?

Nugman, 2024-02-15:
It seems, like the direction of the road is mixed up between the map and the statistics table.

See: https://obs.adfc-hessen.de/map#21.70/50.13958253608688/8.37322566774414
When I select "Richtung Westen", the southern part of the road is highlighted, which leads actually to the east, because we drive on the right hand side. 
[image]

When I select "Richtung Osten", the northern side of the road is selected:
[image]

Is this a bug or just a misunderstanding on my side?

gluap, 2024-02-16:
@Nugman This looks indeed like a bug. I also checked the map but I don't think the segment shape is to blame. (was thinking that the segment was maybe s-shaped making it hard to decide between "west" and "east"-bound.

gluap, 2024-02-20:
Yes, only "osten" / "westen" are the wrong way round. The statistics match the blue line on the map as far as i could tell.

opatut, 2024-07-24:
This is fixed in `next` and will be in 0.9.0.

closed 2024-07-24 by opatut