openbikesensor/portal #364: Enhancement: show streets covered by tracks, even without overtakings

jschirrmacher, 2024-05-01:
I want to know which streets I already covered and which ones not (or not enough) to eventually cover all streets in my region. This would include streets where overtaking didn't happen, because if nobody overtakes me, this is a good sign and valuable to know in statistical analyses.

gluap, 2024-05-01:
@jschirrmacher
When you are logged in on the portal, on the "tracks" page you can select "show only my tracks". When you then select "road usage" as the metric you want to display this seems to be pretty much what you're expecting.

[image]

jschirrmacher, 2024-05-01:
Actually, that's exactly what I wanted!

The trick was the "usage count" - I didn't know that.

closed 2024-05-01 by jschirrmacher