openbikesensor/portal #369: production-deployment does not work

ttjaden, 2024-09-26:
I ordered a new server for a production setup and had the following issue while following your https://github.com/openbikesensor/portal/blob/main/docs/production-deployment.md

`docker-compose up -d keycloak`

results in

`ERROR: pull access denied for jboss/keycloak, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
Attaching to `

because the repository does not exist anymore. I tried only to change `jboss/keycloak:15.1.0` to `keycloak/keycloak:18.0.0` but the container didn't start.

Could you give me a hint, what to do?

ttjaden, 2024-09-26:
Change from `jboss/keycloak:15.1.0` to `quay.io/keycloak/keycloak:15.1.0` works for me. After changing the `docker-compose up -d keycloak` works.

gluap, 2024-09-30:
Glad to see you already found the quay.io keycloak.

Regarding `OBS_POSTGRES_URL` the info you need to set is adding your chosen password (`OBS_POSTGRES_PASSWORD`) instead of `<<TODO>>` in the .env file in the `OBS_POSTGRES_URL` line.

closed 2024-10-07 by ttjaden