openbikesensor/portal #371: Use other csv files other than provided by obs

JohannesSchnell, 2024-10-07:
Hello everyone thanks for sharing your software and also for your good documentation!

I was trying to use your software to visualize data i recorded myself, therefore i picked data format v1.2.

i paid attention to have a BOM:
```
root@d0e454914fc6:/opt/obs/api# xxd -g1 data/input/user/data_obs.csv 
00000000: ef bb bf 4f 42 53 44 61 74 61 46 6f 72 6d 61 74  ...OBSDataFormat
```
i also grabbed the metadata line you provide in your docs, the first lines of my csv look like this:

```
OBSDataFormatVersion=1.2&OBSFirmwareVersion=v0.3.999&DataPerMeasurement=3&MaximumMeasurementsPerLine=60&HandlebarOffsetLeft=30&HandlebarOffsetRight=30&NumberOfDefinedPrivacyAreas=3&PrivacyLevelApplied=AbsolutePrivacy&MaximumValidFlightTimeMicroseconds=18560&DistanceSensorsUsed=HC-SR04/JSN-SR04T&DeviceId=ecec&PresetId=Wade&BluetoothEnabled=1&TrackId=38605ba-76...&TimeZone=GPS
Date;Time;Latitude;Longitude;Confirmed;Left;Right;Course;Speed
2024-07-15;11:16:10.981;51.0;7.0;1;48.0;0;;20
2024-07-15;11:16:15.984;51.1;7.1;1;342.0;0;;20
2024-07-15;16:31:15.924;51.2;7.2;1;137.0;0;;20
2024-07-15;16:31:16.938;51.3;7.3;1;159.0;0;;20
```

this format is accepted during the annotation process yet i think it is still misinterpreted since it says valid & confirmed meas. = 0 as well as not recognizing the timestamps properly(?)
```
root@d0e454914fc6:/opt/obs/api# obs-face -ACV
2024-10-07 08:27:25 root INFO Loading OpenStreetMap data
2024-10-07 08:27:25 obs.bin.obs_face INFO Collecting datasets
2024-10-07 08:27:25 obs.bin.obs_face INFO Annotating and filtering CSV files
2024-10-07 08:27:25 obs.bin.obs_face INFO annotating datasets
2024-10-07 08:27:25 user/data_obs.csv INFO user/data_obs.csv: 39 measurements processed, derived values for 0 measurements (speed: 0, course: 0)
2024-10-07 08:27:25 user/data_obs.csv INFO user/data_obs.csv: Statistics:
total files:            1
total measurements:     39
valid measurements:     0
confirmed measurements: 0
time range:             None to None
continuous time:        0
continuous distance:    0m
continuous segments:    0
Traceback (most recent call last):
  File "/usr/local/bin/obs-face", line 33, in <module>
    sys.exit(load_entry_point('openbikesensor-api', 'console_scripts', 'obs-face')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/obs/api/obs/bin/obs_face.py", line 670, in main
    asyncio.run(amain())
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/opt/obs/api/obs/bin/obs_face.py", line 590, in amain
    measurements, statistics = process_datasets(
                               ^^^^^^^^^^^^^^^^^
  File "/opt/obs/api/obs/bin/obs_face.py", line 188, in process_datasets
    dummy_process.dequeue_and_process()
  File "/opt/obs/api/obs/bin/obs_face.py", line 267, in dequeue_and_process
    measurements = self.annotate(dataset)
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/obs/api/obs/bin/obs_face.py", line 327, in annotate
    measurements = self.measurement_filter.filter(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/obs/api/obs/face/filter/ChainFilter.py", line 34, in filter
    measurements = filter_.filter(measurements, log)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/obs/api/obs/face/filter/RequiredFieldsFilter.py", line 36, in filter
    input_size = len(measurements)
                 ^^^^^^^^^^^^^^^^^
TypeError: object of type 'coroutine' has no len()
sys:1: RuntimeWarning: coroutine 'AnnotateMeasurements.annotate' was never awaited
```

Do you have any hints that can point me towards what I'm trying to achieve?
Thanks in advance!

gluap, 2024-10-07:
Hi Johannes,

I think there might be a misunderstanding. We don't read arbitrary CSV files but only CSV with exactly the columns provided by an OpenBikeSensor and in the order that  an OpenBikeSensor would write them. The reader can read different formats specified by different version specifiers, but always expects the columns required of that file format version.

The list of columns required for the protocol version documented on the [firmware documentation about csv files](https://github.com/openbikesensor/OpenBikeSensorFirmware/blob/main/docs/software/firmware/csv_format.md) can be found at the bottom of that page (note that there is more columns than shown, hence the ellipsis). Your columns differ from the columns that are required in a data format version 1.2 file (which the header of the CSV you posted defines it as). So I guess it's rather surprising that this does not immediately fail upon parsing, which might be (my untested assumption) thanks to the CSV parser assuming default values for the missing columns - But that would of course yield nonsensical data.

Cheers,
Paul

JohannesSchnell, 2024-10-09:
I looked into that repository and spend some time in the debugger to get it to run
you need the following headers for data format version 1.2, order is not important:
```
Date;Time;Latitude;Longitude;Course;Speed;Left;Right;Confirmed
```
Date needs to be in `DD.MM.YYYY`, (distance) Left and Right needs to be of type `Int` in cm.

Thanks again for providing the Code, the annotation process is very complex and I am glad I can reuse yours ;)

gluap, 2024-10-16:
Nice that you were able to solve the issue and thanks for writing down your solution for others.

closed 2024-10-16 by gluap